STMicroelectronics/stm32g0xx-hal-driver #3: Data filters in LL_ADC_REG_ReadConversionDataX prevent left alignment from working

phileimer, 2023-12-26:
**Describe the set-up**
Any board.
Any IDE/compiler.
Noticed with MC Library with HAL/LL that uses ADC value left alignment.

**Describe the bug**
In recent version of HAL/LL for G0, functions LL_ADC_REG_ReadConversionData12, LL_ADC_REG_ReadConversionData10, LL_ADC_REG_ReadConversionData8 and LL_ADC_REG_ReadConversionData6 were modified with data filters, respectively 0x00000FFFUL, 0x000003FFUL, 0x000000FFUL, and 0x0000003FUL.
This prevent them from working when LL_ADC_DATA_ALIGN_LEFT is configured.

**How To Reproduce**
Any project using MC Library or LL_ADC_DATA_ALIGN_LEFT with G0/HAL_LL.

**Additional context**
Removing the filters mentioned above solves the issue.
For example, for 12 bit data valu, the correct function is:
```
__STATIC_INLINE uint16_t LL_ADC_REG_ReadConversionData12(const ADC_TypeDef *ADCx)
{
  return (uint16_t)(READ_BIT(ADCx->DR, ADC_DR_DATA));
}

```

RJMSTM, 2023-12-29:
ST Internal Reference: 169678

RJMSTM, 2024-01-05:
Hello @phileimer,

This not a bug:
the feature "data alignment" shifts data to be aligned on bit 15, therefore data is 16 bits wide [15; 0].
And therefore, to read data, function LL_ADC_REG_ReadConversionData32() must be used.

The other functions ..._ReadConversionData6/8/10/12 are not usable in this mode.

An update description will be made of LL_ADC_REG_ReadConversionData32() to be more generic on use cases.

With regards,
Rania

phileimer, 2024-01-05:
Hi Rania (@RJMSTM),

Thank you for your feedback. I got the point.
I believe you should fill in a bug for this in the X-CUBE-MCSDK project.
In its version 5, this project uses left data alignement and the 12 bit function, triggering a severe bug.
I've not checked with the newer version 6.

Also, I noticed these LL_ADC functions for other STM32 flavors I use (F4 and L0) don't have these data filters implemented yet.
Will they be modified later?

Thank you again,
J. Ph.

FredericVirot, 2024-04-19:
Hello Jean-Philippe (@phileimer),

X-CUBE-MCSDK version 6.x also uses a left alignment configuration for the ADCs, so this issue is present in this line of versions. This indeed causes a severe bug as it makes the MCSDK unusable on G0 with the 1.6.2 version of the G0 LL drivers. 
This issue will be addressed soon in an upcoming version of MCSDK 6.x. 
However, note that X-CUBE-MCSDK version 5 will most probably not be updated to fix this issue. I would strongly advise you to switch to X-CUBE-MCSDK version 6.x.

The other STM32 series are **not** currently touched by this issue.   

With my best regards,

Fred.

RJMSTM, 2025-06-17:
Fixed in : 4c5571be98e1e549e8f710b3fb7ec7aafafc702d

closed 2025-06-17 by RJMSTM