STMicroelectronics/stm32g0xx-hal-driver #4: Timeout parameter in HAL_I2C_Master_Receive/HAL_I2C_Mem_Read not used for I2C_WaitOnFlagUntilTimeout function

Kedrel, 2024-04-22:
Hello,

I found that the Timeout parameter in both HAL_I2C_Master_Receive & HAL_I2C_Mem_Read functions is not used in these two lines: 

https://github.com/STMicroelectronics/stm32g0xx_hal_driver/blob/dabe885cefd3c201c1d8aa1cc98657db7b76a309/Src/stm32g0xx_hal_i2c.c#L1273

https://github.com/STMicroelectronics/stm32g0xx_hal_driver/blob/dabe885cefd3c201c1d8aa1cc98657db7b76a309/Src/stm32g0xx_hal_i2c.c#L2670

I'm not sure if this is a bug, but in my case, I need to change it to : 
_if (I2C_WaitOnFlagUntilTimeout(hi2c, I2C_FLAG_BUSY, SET, Timeout, tickstart) != HAL_OK)_

**How to reproduce the "bug "**: Call the HAL_I2C_Master_Receive or HAL_I2C_Mem_Read function with a timeout parameter of less than 25ms, and don't power up the i2c device to trigger the timeout.

KRASTM, 2024-04-26:
Hello @Kedrel,

Thank you for the report.

The Timeout used in `I2C_WaitOnFlagUntilTimeout` for the 2 functions HAL_I2C_Master_Receive & HAL_I2C_Mem_Read is sufficient to check if the peripheral still busy.

Otherwise, can you share more details about your scenario or project?

With regards.

Kedrel, 2024-04-29:
Hello @KRASTM,

The system i'm working on currently uses a tick of 10 ms to run the whole process. The aim is never to have a task that exceends 10ms so we can handle the "real-time" properly.

When i call "**HAL_I2C_Mem_Read**" using a Timeout parameter at 2 ms, inside of it we have the call of "**I2C_WaitOnFlagUntilTimeout**" which uses a parameter of **I2C_TIMEOUT_BUSY**, which is 25ms. 

So if an I2C peripheral can't respond, instead of loosing 2ms of wait time like the setting i chose, i'll loose 25ms. 

I can't afford to loose 25ms like that (as i have multiple I2C sensors), so i'll change the **I2C_TIMEOUT_BUSY** value to something acceptable for my application. 

With regards.

closed 2024-04-29 by KRASTM