STMicroelectronics/stm32g0xx-hal-driver #5: HAL_SPI_TransmitReceive_IT crashes when TX buffer is not 16 bit aligned

kasjer, 2024-06-14:
**Describe the set-up**
 * Tested on nucleo-g0b1re
 * arm-none-eabi-gcc version 10.3.1 (but compiler does not matter)

**Describe the bug**
Calling function `HAL_SPI_Receive_IT()` or `HAL_SPI_TransmitReceive_IT()` with buffer not aligned to 16 bits crashes MCU

**How To Reproduce**
1. Call `HAL_SPI_Receive_IT()` with unaligned buffer and more then 1 byte to send

**Additional context**
Real problem however is in `SPI_2linesTxISR_8BIT()` function that tries to optimize send using 16 bits access if there is more data.
For **Cortex-M0+** unaligned read results in fault.

```c
static void SPI_2linesTxISR_8BIT(struct __SPI_HandleTypeDef *hspi)
{
  /* Transmit data in packing Bit mode */
  if (hspi->TxXferCount >= 2U &&
    ((uint32_t)(hspi->pTxBuffPtr) & 1) == 0) // <= !!!! Additional check to use 8 bit for unaligned buffer here fixes the problem
  {
    hspi->Instance->DR = *((uint16_t *)hspi->pTxBuffPtr); // <= !!! Crashes here on Cortex-M0+
    hspi->pTxBuffPtr += sizeof(uint16_t);
    hspi->TxXferCount -= 2U;
  }
...
```

KRASTM, 2024-06-19:
ST internal Reference: 184492

KRASTM, 2024-08-15:
Fixed in commit https://github.com/STMicroelectronics/stm32g0xx_hal_driver/commit/1c95709a51b5a32e02c1469e12e5cfe6b8a1ce71

closed 2024-08-15 by KRASTM

kasjer, 2024-08-15:
It's hard do believe that simple bug (unnecessary optimization that was flawed and could be fixed in one line) resulted in change of API that is less functional than before, adds a lot of checks that are not really needed.
It's not that SPI hardware has some limitation that restrict buffer alignment it's just unreasonable approach that uses "optimization".
Now if one uses DMA API that works fine and for some reason wants to switch to IT that has similar API and could be used interchangeable user can face serious crash and go through debugging to find out that both API are not compatible and application has to be aware of this unnecessary limitation (that is not present in non-Cortex-M0 based MCU's).

This is very disappointing solution and hardly could be referred as fixed but if counts toward KPI.