STMicroelectronics/stm32g0xx-hal-driver #6: HAL I2C produces unwanted NACKs in slave mode

lukas-lang, 2024-07-17:
**Describe the set-up**
 * MCU: STM32G0B0RET

**Describe the bug**
It looks like the I2C HAL driver handles the NACK bit incorrectly, at least in slave mode with Slave Byte Control enabled. Consider e.g. the following line:
https://github.com/STMicroelectronics/stm32g0xx_hal_driver/blob/b3638742cfd2e8160f35f616f5ed19877e228f6e/Src/stm32g0xx_hal_i2c.c#L6291-L6292
This tries to "disable address acknowledge" after the current transaction has been completed. Similarly, the code tries to clear the bit in various places. From the documentation however, it is clear that this cannot work:

> Bit 15 **NACK**: NACK generation (slave mode)
The bit is set by software, cleared by hardware when the NACK is sent, or when a STOP 
condition or an Address matched is received, or when PE=0.
0: an ACK is sent after current received byte.
1: a NACK is sent after current received byte.
_Note: Writing ‘0’ to this bit has no effect.
This bit is used in slave mode only: in master receiver mode, NACK is automatically 
generated after last byte preceding STOP or RESTART condition, whatever the NACK 
bit value.
When an overrun occurs in slave receiver NOSTRETCH mode, a NACK is 
automatically generated whatever the NACK bit value.
When hardware PEC checking is enabled (PECBYTE=1), the PEC acknowledge value 
does not depend on the NACK value._
>
> Source: RM0450, Page 762 (rm0454-stm32g0x0-advanced-armbased-32bit-mcus-stmicroelectronics.pdf)

Note the "Writing 0 to this bit has no effect". This means that once the bit is incorrectly set, it can never be fixed again

**How To Reproduce**
1. Set up a board with the MCU, enable one of the I2C peripherals & hook up a device capable of being an I2C master
2. Add this code to the main file of a new project:
  ```c
  char buf;
  
  void start_receive(I2C_HandleTypeDef *hi2c)
  {
	  LL_I2C_SetTransferSize(hi2c->Instance, 0); // set NBYTES to 0 to prevent premature release of clock
	  LL_I2C_EnableReloadMode(hi2c->Instance);
	  HAL_I2C_Slave_Seq_Receive_IT(hi2c, &buf, 1, I2C_FIRST_AND_NEXT_FRAME);
	  LL_I2C_SetTransferSize(hi2c->Instance, 1); // set NBYTES to 1 to release clock & start receiving
  }
  
  void HAL_I2C_AddrCallback(I2C_HandleTypeDef *hi2c, uint8_t TransferDirection, uint16_t AddrMatchCode)
  {
	  LL_I2C_EnableSlaveByteControl(hi2c->Instance);
	  start_receive(hi2c);
  }
  
  void HAL_I2C_ListenCpltCallback(I2C_HandleTypeDef *hi2c)
  {
	  HAL_I2C_EnableListen_IT(hi2c);
  }
  
  void HAL_I2C_SlaveRxCpltCallback(I2C_HandleTypeDef *hi2c)
  {
	  start_receive(hi2c);
  }
  
  /* USER CODE END 0 */
  
  /**
    * @brief  The application entry point.
    * @retval int
    */
  int main(void)
  {
    /* USER CODE BEGIN 1 */
  
    /* USER CODE END 1 */
  
    /* MCU Configuration--------------------------------------------------------*/
  
    /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
    HAL_Init();
  
    /* USER CODE BEGIN Init */
  
    /* USER CODE END Init */
  
    /* Configure the system clock */
    SystemClock_Config();
  
    /* USER CODE BEGIN SysInit */
  
    /* USER CODE END SysInit */
  
    /* Initialize all configured peripherals */
    MX_GPIO_Init();
    MX_I2C2_Init();
    /* USER CODE BEGIN 2 */
  
    HAL_I2C_EnableListen_IT(&hi2c2);
    /* USER CODE END 2 */
  
    /* Infinite loop */
    /* USER CODE BEGIN WHILE */
    while (1)
    {
      /* USER CODE END WHILE */
  
      /* USER CODE BEGIN 3 */
    }
    /* USER CODE END 3 */
  }
  ```
3. This code sets up the I2C slave to continuously receive bytes, e.g.
  [image]
4. Set a breakpoint in the following line:
  https://github.com/STMicroelectronics/stm32g0xx_hal_driver/blob/b3638742cfd2e8160f35f616f5ed19877e228f6e/Src/stm32g0xx_hal_i2c.c#L6291-L6292
5. Initiate two more I2C writes. The first will be acknowledged, then the breakpoint will be hit.
6. Continue execution. The second write will not be acknowledged, because the NACK bit was set after the next byte already started being received:
  [image]

**Additional context**
At least in my use case with Slave Byte Control enabled, this leads to spurious NACKs if you're unlucky with timing (see reproduction steps). Simply removing the lines appears to work in my case, and I don't see what that could break. Am I missing something?

lukas-lang, 2024-07-19:
Additional note: At least from my tests, the address byte cannot even be NACK'ed in the first place: Even with clock stretching & SBC enabled, the NACK bit is ignored when releasing the clock after the address byte. That is, the address byte will always be ACK'ed (if it matches the configured address of course).
This means any attempts at "disabling address acknowledge" are even more likely to be incorrect.

KRASTM, 2024-07-19:
Hello @lukas-lang,

Thank you for the report.

If I'm not wrong, you are using a custom board with this µc `STM32G0B0RET`. In order to reproduce the issue, could you please test your project on our board such as **NUCLEO-G071RB** or share an extract of your project (at least the config). we need to know also the version of HAL driver.

In the meanwhile, I will try to analyze further your issue.

With regards.

lukas-lang, 2024-07-21:
Unfortunately, I don't have a NUCLEO-board to test with - the board I am testing with is extremely barebones though, so it should be easy enough to reproduce the issue on something else: Apart from a the recommended decoupling capacitors, it only has two pullups for the I2C lines, & all the relevant pins are simply connected to some headers. You can find the entire code here: [I2C_NACK.zip](https://github.com/user-attachments/files/16324418/I2C_NACK.zip) (it's an almost empty project with just the bare minimum set up to reproduce the issue).
As an I2C master, I am using an [Analog Discovery 3](https://digilent.com/shop/analog-discovery-3/), but of course any other type of logic analyzer or similar should be fine. I set it up to provide 3.3V to the MCU & configure the pins for the I2C bus. Then I simply write some random bytes to the MCU (I2C address 0x01) & verify that they are ACK'ed, as described above. Setting the breakpoint & starting two writes in a row however leads to the second write being NACK'ed as described.
I hope this helps

KRASTM, 2024-07-22:
Hello @lukas-lang,

Thank you for the example. I will try to reproduce the issue using it.
I noticed this Line of code `LL_I2C_SetTransferSize(hi2c->Instance, 1); // set NBYTES to 1 to release clock & start receiving`

The **TransferSize** is "The number of bytes to be transferred", did you intentionally set it to 1?

With regards,

KRASTM, 2024-07-30:
Hello @lukas-lang,

Regarding your issue, I have checked with our team that this line of code is not supposed to cause problem, and they aware there is the possibility to remove it, but not in the near future.

In the other hand, you can remove this line of code if it suits your application.

Please, allow me to close the issue, thank you for your comprehension.

With regards,

closed 2024-08-02 by KRASTM